nodemcu/nodemcu-flasher #3: The 64 bit version posted on 2014-12-30 doesn't work for me

dshiffman, 2015-01-04:
flashing with the default configuration goes OK but then the ESP module won't reboot to the prompt. Changing the configuration to internal://NODEMCU @ 0x00000 causes the 20141209 version to be flashed.
Has anyone gotten the 20141230 flasher to work?

pipi61, 2015-01-04:
1230 working:
NodeMcu 0.9.4 build 20141230  powered by Lua 5.1.4
lua: cannot open init.lua
Whats version flashed? from prebuilt directory...?

dshiffman, 2015-01-04:
Thanks for the response.  Version is the current one at https://github.com/nodemcu/nodemcu-flasher/tree/master/Win64/Releasebuilt 4 days ago which I think is 20141230.
The version that gets loaded if you change the configuration to internal://NODEMCU @ 0x00000 is  "NodeMcu 0.9.2 build 20141209  powered by Lua 5.1.4" and works fine.
By the way, where is the modified configuration stored on a windows machine?  I can't get the flasher to revert to the default settings.
Thanks
Dan

Date: Sat, 3 Jan 2015 17:42:38 -0800

vowstar, 2015-01-05:
I'm so sorry for that 20141230 flasher have mistaken addresses.
INTERNAL://NODEMCU now is fallback mode.
The updated version have fixed this bug, and added an option to restore default settings.
The settings file is under the folder named Config (At the same level of application) and it is Config.xml.
Add the QR code with history MAC also in this folder.

dshiffman, 2015-01-05:
Thanks so much! That fixed it!

Date: Sun, 4 Jan 2015 19:24:40 -0800

closed 2015-01-05 by vowstar