nodemcu/nodemcu-flasher #4: Get these running on a mac?

sticilface, 2015-01-21:
Any releases for mac, or some instructions for how to build them?

I've never done that so directions would be required!

vowstar, 2015-01-22:
This software currently use VCL framework, and VCL is only for Windows. We are developing another version use QT that can build and run under MAC OS.

sticilface, 2015-01-22:
great thanks, is there an ETA?  days - weeks - months - or longer ?

jeffhuys, 2015-02-17:
I'd say months or even longer. These guys don't really take this the right way. They should've created a simple command-line utility instead of a GUI at first, because a GUI /should/ only be an interface for the command-line utility.

vowstar, 2015-02-18:
I'm so sorry that I spent the whole time on how to make the firmware stable. I'm very want to make a good flash tool that cross-platform, sorry for my slow coding.

crapthings, 2015-03-18:
bump +1

TerryE, 2015-03-29:
I realise that most users run windows, but you guys are trying to sell into a niche market. Amongst this hacker user community, the proportion of Linux and Mac users is a lot higher. Using QT is well and good, but it's also quite complex. As Jeff Hughes says, a simple contentions variant would be a better first step for this set of users.

Even so, I really like what you guys are doing thanks.

marcelstoer, 2015-05-16:
In general I do like to have choices and options but in this particular case I don't understand why @vowstar / @nodemcu don't join forces with https://github.com/themadinventor/esptool. The Python tool works quite well (except for the new NodeMCU dev kit, see https://github.com/themadinventor/esptool/issues/40). If there really should be a GUI tool then it could be built on top of `esptool.py`.
I'd even argue that if you're uncomfortable using command line tools you probably won't get very far in your IoT/ESP8266 endeavors but that's of course just my2cents.

vowstar, 2015-05-17:
@marcelstoer  esptool.py is a greate project and we are already using it in our project, and it is recommend for flash nodemcu.  themadinventor/esptool#40 is my mistake, I will test it and try close it. I have a forked version https://github.com/vowstar/esptool support auto download for the new NodeMCU dev kit, GUI tool built on top of esptool.py is a good idea, we are finding a way make it easy to use. This project nodemcu-flasher can't run on OS X, so we will stop develop it, and make a new GUI tool instead. May be GUI based on esptool.py is a nice choice.

marcoskirsch, 2015-07-06:
I agree with @marcelstoer. People using NodeMCU are most likely comfortable with a command line utility and probably even prefer it! Making a nice GUI is _a lot_ of effort that could be invested elsewhere. Making a nice cross-platform GUI using Java or Qt is nearly impossible! (See: ESPlorer).

TerryE, 2015-07-06:
+1.  A GUI tool is a nice to have for newbies in general, but the ESP8266 application developers are mostly hackers to begin with.  The thing about a command is that you can script it.  It's also a *lot* easier to  read (and if necessary tweak) the source of command line tools.

We just need to be sure that the scripting language that they are implemented in has good cross-platform support (e.g. Linux, OSX and WinXX).

phodal, 2015-08-10:
Hi  @TerryE @sticilface @jeffhuys @marcoskirsch ,
  
I try to write such a editor called [iot-dev](https://github.com/iot-works/iot-dev) base on https://github.com/spark/spark-dev & https://github.com/atom/atom, but now I have some trouble on rewrite esptool in Node.js which called [esptool.js](http://github.com/phodal/esptool.js)