nodemcu/nodemcu-flasher #9: Needs some kind of improved progress indicator

uchobby, 2015-05-10:
I have used a prior version named nodemcu_flasher with success, it had a progress bar that seemed to work fine.

Today, with the latest version, with a new name, for some reason...  I don't see a progress bar, have been watching the little icon thing turning slowly as can see serial traffic is occurring but no progress.  Its not done, so maybe something strange is going on.  Has been 10 minutes of TX and RX LED flashing with no indication of progress...

uchobby, 2015-05-14:
Progress Bar not functional.

After starting Log shows ...
Note:Detect serial port changed.
Note:Auto MAP serial port.Port-->COM5

Note:Serial port connected.
Note:Begin find ESP8266.
Note:ESP8266 ACK success.

No indication of progress. The little Nodemcu Team icon in the lower left is turning slowly.

emcniece, 2015-07-19:
Mine was getting stuck at `Note: Being find ESP8266.` but it looks like it might have just been a bad chip? Swapped it out for another and the log is reporting `Note:Program flash success.` now.