nodemcu/nodemcu-flasher #10: Linux support

ge0rgecz, 2015-06-04:
could you support Linux OS? This operating system is popular in DIY comunity ... (arduino developement is perfectly supported on linux platform).

vowstar, 2015-06-07:
Yes, we intend to do so . I am a Linux enthusiast.

omerk, 2015-07-08:
@vowstar Any updates on the Linux support?

ge0rgecz, 2015-07-16:
good news, thank you.

sheepbao, 2016-03-16:
@vowstar could you support Linux OS right now?

brynnb, 2016-07-17:
Just a +1 that Linux support would be great. Currently working on getting an ESP8266 working with my RPi since I'm still waiting on USB/TTL converter in the mail. However, esptool doesn't work with the ESP-12-E and seemingly the only known, documented method to make it work is to flash NodeMCU with the NodeMCU-flasher tool.

johndoe71rus, 2017-04-12:
hi. almost 2 year pass. Now, what about linux support?

johndoe71rus, 2017-04-15:
https://github.com/marcelstoer/nodemcu-pyflasher
it work GUI for Linux

Sinan81, 2018-10-21:
+1 for Linux support.