nodemcu/nodemcu-flasher #13: Trying to flash two files one after the other

nkolban, 2015-09-26:
I have a solution that needs to upload two binaries into flash.  I have configured the tool as shown in the image:

[image]

However, when I flash, only ONE file is uploaded.  It appears I can only select ONE file at a time ... I had assumed that I could select all files and they would process.  Is this not correct?

nissimzur, 2015-09-26:
Nkolban,
I can check that from the source code, but my Delfi is very old. What version of Defei this is ? (16. 17.2 ..)
And where I can downloaded it ? 
-Nissim

nkolban, 2015-09-26:
Howdy Nissim,
I'm afraid I don't understand.   What is "Delfi" or "Defei"?  If you are asking where I got the binary ... it was from the Win64/Release folder found here:

https://github.com/nodemcu/nodemcu-flasher

nissimzur, 2015-09-26:
Hi Neil,

I misspelled Delphi. I like to compile and check where the bug arrived, Since old version was OK. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email       <mailto:nissim@vitelix.com> nissim@vitelix.com  

 <http://www.vitelix.com/> ViTELiX |   <http://www.usbphone.com/> USBphone |  <http://www.elinistech.com/> Elinistech

 

https://github.com/nodemcu/nodemcu-flasher

nkolban, 2015-09-26:
Let me know if there is anything I can do to assist.

nissimzur, 2015-09-26:
Do you have a Delphi license ? 

I had a trial version that expire. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email       <mailto:nissim@vitelix.com> nissim@vitelix.com  

 <http://www.vitelix.com/> ViTELiX |   <http://www.usbphone.com/> USBphone |  <http://www.elinistech.com/> Elinistech

nkolban, 2015-09-26:
Howdy ... I'm afraid I don't know what "Delphi" is.   I am just an end user who was hoping to use the tool so I downloaded the .exe file which ran and gave the issue reported.   Beyond that, there isn't much more I can offer other than to collect any logs or demo the issue.

nissimzur, 2015-09-27:
Delphi is the programming language of this application. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email       <mailto:nissim@vitelix.com> nissim@vitelix.com  

 <http://www.vitelix.com/> ViTELiX |   <http://www.usbphone.com/> USBphone |  <http://www.elinistech.com/> Elinistech

vowstar, 2015-09-29:
Please try latest version, thanks

nkolban, 2015-09-29:
I downloaded the latest version (2015-09-29) and tested with multiple files.   My tests now shows that if I select two files for flashing, they are indeed both flashed.

closed 2015-09-29 by nkolban

nissimzur, 2015-09-29:
OK. Thank you.

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email       <mailto:nissim@vitelix.com> nissim@vitelix.com  

 <http://www.vitelix.com/> ViTELiX |   <http://www.usbphone.com/> USBphone |  <http://www.elinistech.com/> Elinistech

vowstar, 2015-09-30:
Please try latest version and support new OUI of esp8266 chips.
Some new chip's MAC have changed:
Old OUI:
        18-FE-34
New OUI:
        5C-CF-7F