nodemcu/nodemcu-flasher #16: Unresponsive modules after successful flash

mamert, 2015-11-11:
I can't identify the reason.
Steps I take:
1. Launch /Win64/Release/ESP8266Flasher.exe (2015-09-30) on Windows 8
2. Leave default settings (just INTERNAL://NODEMCU)
3. GPIO0 low, click FLASH, wait till done

Known symptoms afterwards:
Module is connectable (COM3 or 4), but does not respond - e.g. trying to flash again doesn't even obtain MAC; LuaFlasher (GPIO0 disconnected) shows no responses.

This has already affected: 1*ESP-01, 4*ESP-07, 1*ESP-12 via 3.3(actually 3.6)V USB2Serial adapter I use for Arduino Pro Minis, and (tried just to be sure) a verified working nodemcu dev board via USB.

Please instruct me on what to do to obtain additional information.
I still have: 1*ESP-07, 1*ESP-03, 2*ESP-05(not flashable).

mamert, 2015-11-12:
Update: The devkit and ESP-01 have been repeatedly flashed without incident, after this. The others have not recovered.
While each of these modules has one "successful" flash with nodemcu-flasher behind it, the program likely has nothing to do with the problem, after all. Probably some hardware failure or quirk. Sorry.

pgScorpio, 2015-11-12:
Try flashing with unused gpio pins tied to vcc (or very strong pullup <1k) !
I know some of them causes problems sometimes, and esp-01 has tied them hard to vcc...

mamert, 2015-11-12:
It does? Which pins? I've seen tutorials on breaking out GPIO 4,5,12-15, so it can't be those.
On ESP-07 and ESP-03, That leaves GPIO16.
Connecting GPIO16 to VCC:
2xESP-07 still show "failed to connect". 1x ESP-12  - well, progress there (or just luck): No trouble connecting, now it shows either "invalid head of packet" or "failed to write to target Flash" at 0%. So, maybe broken memory, or it doesn't like my FT232RL.

pgScorpio, 2015-11-14:
All I know for shure to reliably start  programming (always worked for me):
GPIO 0 -> GND
GPIO 2 -> VCC
GPIO 15 -> GND
CH_PD (CH_EN) -> VCC
RESET -> 1.5k to 10k pullup to VCC
Apply power.
Pull reset low.
Start dowload and immediately release reset.

From hearing: Other floating GPIO pins CAN cause problems entering download mode when ESP has non-standard firmware (using strong pullups solves this problem most of the time). Some special firmwares may also require a special GPIO pin to be connected to GND, check firmware documentation.

The only problems I've encountered is that flashing back "stock" firmware no longer works after another (lua) firmware has been flashed... (Firmware starts but Auto resets on most AT commands)
In these cases (all ESP7 or 12 variants) I could still flash the other (lua) firmware again and that one worked fine again...

Verzonden vanaf Samsung Mobile

https://github.com/nodemcu/nodemcu-flasher/issues/16#issuecomment-156267511

pgScorpio, 2015-11-14:
To Mamert:  
"failed to write to target Flash"
Is most likely the GPIO 15 -> GND..
It is possibly trying to write to a sd card...

mamert, 2015-11-15:
Thank you, pgScorpio! At some point I forgot and ended up using the connections as for ESP-01, which did not use GPIO15. Mea culpa.
2 of the first modules I've tried still exhibit the original problem - but further testing, now that I can flash them again (ESP8266 Flash Downloader), reveals them to be currently unresponsive regardless of firmware, so it has nothing to do with this software, after all.

closed 2015-11-15 by mamert