nodemcu/nodemcu-flasher #17: Mac support

eallik, 2015-11-18:
Can't say Macs are common in the DIY landscape (maybe they are maybe no, I don't know) but I'm on one so I'm opening this ticket to propose adding OS X support, in addition to Windows, and the already I guess planned Linux support.

Is this something one could hope for? Or completely out of the scope of this project?

marcelstoer, 2015-11-20:
The README has been saying
> We are working on next version and will use QT framework. It will be cross platform and open source.

for nearly a year now :wink:. So, no, not completely out of the scope.

eallik, 2015-11-21:
Aha, that's both good and slightly bad news at the same time, but more good I think :)

pnewb, 2016-01-09:
Macs are pretty common out here.  Count this as one more vote for bumping priority on mac support.

w1jp, 2016-02-08:
Macs are actually good hacker platforms in many ways better than Win. It is UNIX underneath so all sorts of open source stuff works on it. It natively supports connecting to DCE/DTEs (via 'screen') really easy. 

So I second the supporting of Mac... Where is the Qt version?

pilhuhn, 2016-02-18:
Did anyone try to use some Pascal for OSX (if such a thing exists)?

marcelstoer, 2016-02-18:
I for myself don't miss the OS X support as I use esptool.py (on Mac). Side note: on http://nodemcu-build.com/ I see 68% Win, 13% Linux and 12% OS X.

pilhuhn, 2016-02-18:
On 18 Feb 2016, at 8:40, Marcel Stör wrote:

> I for myself don't miss the OS X support as I use esptool.py (on Mac). 
> Side note: on http://nodemcu-build.com/ I see 68% Win, 13% Linux and 
> 12% OS X.

I am using esptool myself - but apparently the flasher knows all the
secrets of burning additional .bin files at various addresses like
hinted at: 
https://github.com/nodemcu/nodemcu-firmware#files-to-burn-to-the-flash

"  Also, in some special circumstances, you may need to flash blank.bin 
or esp_init_data_default.bin to various addresses on the flash 
(depending on flash size and type). "

And I've read that to go from 0.9.5 to something > 1.0 you have to flash 
those .bin files

marcelstoer, 2016-02-18:
Yep, I look forward to the firmware `dev` branch being merged back to `master` because we've got much better documentation now: http://nodemcu.readthedocs.org/en/dev/en/flash/

pilhuhn, 2016-02-18:
On 18 Feb 2016, at 8:49, Marcel Stör wrote:

Yes, 
http://nodemcu.readthedocs.org/en/dev/en/flash/#upgrading-from-sdk-09x-firmware
is the other reference - which talks about "depending on flash size".

I just saw revisiting my Huzzah ESP, that this has 4MB of flash
according to `node.info()` So I guess it may be an ESP-12E
internally.

Robiv8, 2019-02-16:
Nov 2015 Today Feb 2019
Any news about the OS X support

marcelstoer, 2019-02-16:
Just use https://github.com/marcelstoer/nodemcu-pyflasher (from yours truly)