nodemcu/nodemcu-flasher #20: Tool doesn't work on win10-64bit

MishraShivendra, 2016-02-21:
I tried to run this tool on windows 10 - 64 and it doesn't run. Sometimes, It gives an error "App is incompatible."

sihui62, 2016-03-09:
Cannot confirm that: works fine on my W10 64Bit.

gourneau, 2016-03-11:
As of today the tool worked fine for me with Windows 10 64bit.

gtsoftwareltd, 2016-05-31:
I just tried this on Windows 10 64-bit and I can confirm that the flasher does work.

One thing I have seen is that people just download the flasher exe and end up with a small file (about 44KB) that is actually a HTML page. This can happen by right clicking on the exe and choosing to 'save link as'; this isn't what you want. Download the whole Zip (or clone) and then run the 6MB exe.

bubba198, 2016-07-11:
r-click and "run as administrator" -- that's at least my experience since my user account apparently doesn't have local administrator rights (Microsoft term lol)

Sygyzmundovych, 2016-11-06:
[image]
Got this on Win10 64bit running as Administrator

closed 2016-11-13 by MishraShivendra

Imperativ, 2020-01-10:
> One thing I have seen is that people just download the flasher exe and end up with a small file (about 44KB) that is actually a HTML page. This can happen by right clicking on the exe and choosing [...]

It's exactly that. Thanks for the tipp. Maybe someone could publish this info more prominent...and not in a small threat under closed issues...