nodemcu/nodemcu-flasher #22: Load init.lua with flashing

danlu01, 2016-04-15:
Is there a way to load init.lua file with the firmware flashing tool when flashing new firmware?

nissimzur, 2016-04-16:
Since the file system aria  is not part of the firmware, I presume it passable after ending the flashing cycle with reset in the end to run a code for file loading to load   init.lua in the same tool. 

This seems important with update device installed already at user side. 

 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

marcelstoer, 2016-08-21:
You can do that with `spiffsimg` but you need Linux to run it: http://nodemcu.readthedocs.io/en/latest/en/spiffs/.