nodemcu/nodemcu-flasher #25: set GPIO0 to LOW

pepeEL, 2016-09-22:
Hi
I have from aliexpress clone nodeMCU V3. But i can not programming it under arduino IDE

Szkic używa 243 785 bajtów z (23%) pamięci programu. Maksimum to 1 044 464 bajtów.
Globalne zmienne używają 34 468 bajtów z (42%) dynamicznej pamięci, pozostawiając 47 452 bajtów dla lokalnych zmiennch. Maksimum to  81 920 bajtów.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

and also can not flash firmware by this flsher...
I dont know how set GPIO0 to LOW and where i can found latest firmware 1.5.4.1-master_20160802 in bin file ? https://github.com/nodemcu/nodemcu-firmware/releases
In bin file is only available version 0.9.6-dev_20150704: Fix documents.

Please help me...

nissimzur, 2016-09-22:
Call my Skype nissim.test

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <https://goo.gl/CXY35s> Linkedin  https://goo.gl/CXY35s
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

pepeEL, 2016-09-22:
But i dont speak good english. Very easy with translator
Can you help me on this post...?