nodemcu/nodemcu-flasher #27: FLASH DOES NOT WORK!

eholl, 2017-04-30:
When i press flash it does no download anything the bar does not even come up please help me

amrithmmh, 2018-07-20:
same ....not flashing

kratos13, 2018-09-24:
I have been working for over 12 hours trying to flash this. I have tried with Windows, Kali, Debian, and OS X.  ONE time it was successful, but then bricked (again) after loading a payload and rebooting the device. This is becoming mentally exhausting. I have performed the steps so many times, I've memorized them. I upload, the export, then flash, then upload again. Device never resets itself... ANY help would be appreciated.

PlastiBots, 2019-10-09:
Hi Guys.. FYI I got it working.  Ive got a clone NodeMCU mini and was fighting with the same issue. It would just flash the blue LED and nothing would happen.  Here's what worked for me. 1) make sure defaults are selected under the Advanced menu.  Then before plugging the Node into USB, I held down the Flash button, plugged it into USB cable, released the flash button, then clicked the flash.  Only then did it ID the MAC addresses and start flashing.  Now I just need to verify if it works..  Not done this before...