nodemcu/nodemcu-flasher #28: password reset

willikynah, 2017-05-11:
Hey,
i've recently changed the password on the device, but now accidently forgot it.... How do i reset it though ?

nissimzur, 2017-05-11:
Flash Firmware again, and fill all zero before. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <https://goo.gl/CXY35s> Linkedin  https://goo.gl/CXY35s
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

HuskyArts, 2017-05-13:
Hello Sir 
The exact sam thing as stated previsously at the top of this thread has happend to me
I have done what ive been told and re-flashed my NodeMCU however this changed nothing it apears to rember that it one had a name and a password. if you can help me that would be greatly apreciated.

nissimzur, 2017-05-13:
If you fill all flash with zeros it can’t remember anything from old data. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <https://goo.gl/CXY35s> Linkedin  https://goo.gl/CXY35s
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com