nodemcu/nodemcu-flasher #29: Nodemcu isn't detected (COM)

n00bhax, 2017-05-19:
the program doesn't see the com and in device manager it's the same. Nothing is connected. I do have an arduino nano which works perfectly and is detected by device manager. I tried 2 different cables and pressing the flash and reset buttons on the esp8266.

Thanks for helping!

celgara, 2017-10-18:
So most likely the reason for this is a driver issue the driver is cp2102 driver esp8266
 or here is a link hope this helps
 https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers 

just unzip the driver and run the exe and you should start seeing the esp8266

n00bhax, 2017-10-18:
I don't think this will work. The nodemcu was just broken. I bought another one and it worked fine

n00bhax, 2017-10-18:
thanks for the answer though

celgara, 2017-10-18:
no problem

mlinaje, 2017-12-07:
I guess issues should be close since it is not a nodemcu problem, but a USB-to-Serial IC problem.
Look at https://github.com/nodemcu/nodemcu-flasher/issues/31
Close issue 29 and issue 31 :-) 👍

tristanlirette, 2020-06-22:
> 
> 
> So most likely the reason for this is a driver issue the driver is cp2102 driver esp8266
> or here is a link hope this helps
> https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
> 
> just unzip the driver and run the exe and you should start seeing the esp8266

Worked for me, the COM port was available after installing the driver.