nodemcu/nodemcu-flasher #30: AP and STA MAC addresses shown are wrong

gotfredsen, 2017-06-17:
The last byte of the MAC addresses shown is wrong. In my case it should be `0x48` and it shows `0x12`.
If I modify line `884` in the [UnitFormMain.pas](https://github.com/nodemcu/nodemcu-flasher/blob/master/UnitFormMain.pas) from
`MACAddr[5] := ((MACReg0 shr 24) And $FF);`
to
`MACAddr[5] := ((MACReg0 shr 22) And $FF);`
it will work.
Please confirm if that indeed is correct.
Thanks.

iFish85, 2017-06-30:
My solution is like this:
modify line 827 in the UnitFormMain.pas from
RawByte := LeftStr(Frame, Length(Frame) - Position + 1);
to
RawByte := MidStr(Frame, Position, Length(ESP_READ_REG_ACK));

gotfredsen, 2017-07-01:
Thanks @iFish85, that worked much better.

closed 2017-07-01 by gotfredsen