nodemcu/nodemcu-flasher #31: NodeMcu 8266 Clone not detecting on PC

TrapFather, 2017-06-30:
I have being trying to plug in my NodeMcu 8266 Clone from China but it's not being detected by my PC
I have tried numerous different cords (including data transfer cables) and installed all the CH340G Drivers, ESP8266 Drivers and had no luck!

If anyone has a solution please help me.
I appreciate your help.

Regards,
TrapFather

MBartner, 2017-08-18:
I had the same problem, but all it took for me was opening up device manager and clicking on the Ports(COM & LPT) drop down for it to detect. Hope that works.

yonghuming, 2017-09-06:
@TrapFather do you use windows or ubuntu?

mlinaje, 2017-12-07:
In my case the undetected board is a lolin v3. Trying to discover if the problem is the ESP8266 or the USB-to-Serial converter, I discovered that at least in my case was the later.
I was able to flash a new nodemcu firmware using the nodemcu flasher tool while overcoming (i.e., not using) the CH430 USB-to-Serial converter onboard and replacing it using a regular FDTI USB-to-serial converter. 
FDTI  <----->   ESP8266 nodemcu
-------------------------------------
  RX                                TX
  TX                                RX
GRND                          GRND
GRND                           D3
  3.3V                            D4

It worked and even I haven't try it, I guess programming the board (e.g. lua or arduino) is possible using the same configuration.
Best

dzp0wer, 2019-01-17:
I cant find  "a regular FDTI USB-to-serial converter",  do you have precise references pls

mlinaje, 2019-01-18:
> 
> 
> I cant find "a regular FDTI USB-to-serial converter", do you have the precise references of your converter pls

@dzp0wer like this one among others: https://www.banggood.com/FT232RL-FTDI-USB-To-TTL-Serial-Converter-Adapter-Module-For-Arduino-p-917226.html?rmmds=search&cur_warehouse=CN

Just remember to buy one working at 3.3V as ESP8266 does. In the one I linked just remember to put the jumper in the 3.3V position.
Best,