nodemcu/nodemcu-flasher #36: Wemos NodeMCU v3 Lua clone dev usb board won't flash

soydemadrid, 2018-05-20:
Hi I am trying to flash a wemos nodemcu esp12 dev board via usb and the nodemcu-flasher but I just get stuck on the log at: Note:Begin find ESP8266.

The blue com light just keeps flashing and nothing happens...

On the Operation tab there is a rotating message saying require("gpio") and require("wifi") that just flash round over and over.

Any help to be able to flash a firmware on this board would be good, I have tried 3 of the same dev boards and none will flash but I know they work as I can flash them via Arduino IDE and connect to them ok and upload blink demo etc

Kirakodo, 2018-08-31:
You only must set to the defaults

sumitkanu, 2019-05-27:
Hey
I am facing similar problem
I am using LOLin nodeMCUv3 and while using the firmware programmer it gets stuck on
Note: Serial port connected.
Note: Begin find ESP8266.
Note: ESP8266 ACK success.
Note: ESP8266 ACK success.
Error: Set ESP8266 Address timeout.
Settings are the default. This problem started after I started uploading different code while one was still uploading using ide. Do you know how I can solve this issue, I tried all sort of flashing tools and nothing works.
in Arduino ide when I try to upload a blank sketch, I get this error:
esptool.FatalError: Timed out waiting for packet header

Tech-TX, 2019-11-23:
One point not mentioned above is flash brand and size.  I have a Wemos NodeMCU V3 with an XMC 32Mb flash, part number XM25QH32B.  There are two issues in the esp8266/Arduino project that are quite likely relevant to this issue:
https://github.com/esp8266/Arduino/pull/6552
https://github.com/esp8266/Arduino/issues/6559
In the first, they lower the flash frequency to 26MHz to get reliable operation with the default 75% output driver strength of these parts.  In the second it appears they patch the boot to set them to 100% at reset.  To get reliable operation at 40MHz or faster, the drive level must be 100%, and yet the two control bits are volatile (not maintained across a power cycle).  Even if you can program it, it may not run reliably unless you set the flash speed to 26MHz.
[image]
[image]