nodemcu/nodemcu-flasher #37: GPIO low

ojvillar, 2018-07-07:
how to make nodemcu esp8266 GPIO LOW

NoTrace-Kai, 2018-10-03:
Hey there ...
got the same problem ... i set the output D1 and D2 to low, but when i boot my NodeMCU
this GPIO Pins are high. I can reproduce that also with D3 and D4 ... when i edit the GPIOs in
my config file.
What can i do?

peter-hartmann, 2019-12-01:
The readme file is really not describing it well, please fix.

Basically when the esp boots up i.e. power on or reset, then it reads the GPIO-00 pin.  If the pin is LOW then the esp goes into listening mode awaiting a firmware update.
That all happens before any of your programs run.  This can only be done with a wire connected to the GPIO-00 pin bridging it to Ground on boot or reset.

1. Connect a wire bridging GPIO-00 to ground (on a wemos board this is pin D3)
1. While the wire is connected power the esp on on or reset it, either way.
1. Now click the Flash button in the nodemcu-flasher and watch the progress bar as your esp is getting its firmware updated.

Hope this helps.