nodemcu/nodemcu-flasher #38: No COM showing up in ESP8266Flasher

dhylde17, 2018-12-24:
Hello 
I'm trying to flash but no COM is showing up in the app.
here's what I've done :
- I've tried changing USB cable (half a dozen, i didnt know I had so many sigh)
- I've installed the following drivers for my Windows 10, https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers 
- In Windows it is showing up has COM 4
- In the Arduino IDE I can see the COM 4 but not in ESP8266Flasher

Can you help with the issue? Any tips on how to flash?
thanks for the help

dagnall53, 2019-03-15:
Same experience: No selectable Com port..

dagnall53, 2019-03-18:
Looks like it works if you actually type COM4 in the select box.. (NO gap) 
Hope this helps. Its a nice tool otherwise!
- and you have to physically press reset on the board afterwards.. --