nodemcu/nodemcu-flasher #45: after flashing NodeMCU is not recognize by Windows anymore

Mrivilla, 2019-12-09:
Hi Dears,
I flashed my NodeMCU with the tool. I connected my NodeMCU to Windows and COM6 was recognized. 
After flash the NodeMCU to appli the new firmware (by default), the NodeMCU is not working anymore, so Windows does not recognize the USB devise, so cannot be linked to COM6.

Any suggestion to flash or recover the NodeMCU?

Thanks.
Marcos.

Raidr1212, 2019-12-12:
I have the exact same problem

nissimzur, 2019-12-12:
Com port has nothing with the esp chip. Mind that pcb gnd must be connect
to pc gnd. Com port need rx tx gnd

Raidr1212, 2019-12-12:
> Com port has nothing with the esp chip. Mind that pcb gnd must be connect to pc gnd. Com port need rx tx gnd בתאריך יום ה׳, 12 בדצמ׳ 2019, 16:02, מאת Raidr1212 ‏< notifications@github.com>:
> […](#)

I am talking about the nodemcu micro USB port I'm not using wires I'm using a regular data phone cable

nissimzur, 2019-12-12:
The usb expos com port to pc. Disconnect from pcb. Sort Rx to Tx. Entre to
terminal app and type and see if you got back the character

Raidr1212, 2019-12-12:
> The usb expos com port to pc. Disconnect from pcb. Sort Rx to Tx. Entre to terminal app and type and see if you got back the character בתאריך יום ה׳, 12 בדצמ׳ 2019, 16:32, מאת Raidr1212 ‏< notifications@github.com>:
> […](#)

[image]
I'm not quite sure you know what a nodemcu is

wacher74, 2019-12-14:
I experienced same but reset to default is helped me. (baud rate jump to 230400 )

Raidr1212, 2019-12-14:
> I experienced same but reset to default is helped me. (baud rate jump to 230400 )
Can you help me on doing that without having an com port?